espressif/esp-mesh-lite #177: Is it possible to freeze the tree topology or mesh parent?

mrjohannchang, 2025-06-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

In our testing, we observed that automatic mesh formation is not always fast enough and occasionally results in loops or unreachable (dead-end) devices. Since we already know the physical locations of all devices, I would like to freeze the mesh topology once the devices are deployed, in order to make the network more deterministic and reliable.

I am considering two potential approaches to accomplish this:

**Option 1: Fix Mesh Layers Using** `esp_mesh_lite_set_allowed_level()`

Set each device to a specific mesh layer using `esp_mesh_lite_set_allowed_level()`.

* **Question 1.1:**
    If a device is set to level 3 using `esp_mesh_lite_set_allowed_level(3)`, but no parent node at level 2 is available, it appears that the device will not be able to join the mesh.
     
    * If I’d like to decide whether to fall back to automatic mesh formation based on actual connection attempts rather than a static timer. Is there a way to track the number of join attempts or retry attempts made by esp-mesh-lite?

* **Question 1.2:**
    Since `esp_mesh_lite_set_allowed_level()` must be called **before** `esp_mesh_lite_start()`, how can I reset or reconfigure the allowed level **after** startup?

    * Is there a supported way to stop and restart the mesh stack, such as a esp_mesh_lite_stop() function or equivalent?

**Option 2: Fix Parent Node Using** `esp_mesh_lite_set_wifi_config()`

Use `esp_mesh_lite_set_wifi_config()` to force a device to only connect to a specific parent.

* **Question 2.1:**
    Is there a way to lock a node to a specific parent using this function?

    * There doesn’t appear to be any documentation for `esp_mesh_lite_set_wifi_config()`. Could you please provide an example of how to use this function to assign a fixed parent?

* **Question 2.2:**
    If the specified parent becomes unavailable and I want to fall back to automatic mesh formation:

    * What is the proper way to reset the configuration and rejoin the mesh? Could you also provide an example for this fallback behavior?

Thank you in advance for your guidance. This information will help us achieve a more stable and controlled mesh deployment. Looking forward to your response.

mrjohannchang, 2025-06-16:
If I call these APIs

```
esp_mesh_lite_disconnect()  
esp_mesh_lite_set_wifi_config()  
esp_wifi_connect()
```

after `esp_mesh_lite_start()` to manually connect to a specific device, will the internal mesh connection loop be stopped by `esp_mesh_lite_disconnect()`? If not, after configuring the parent info via `esp_mesh_lite_set_wifi_config()`, the configured parent info can still be flushed by the internal mesh loop right?

tswen, 2025-06-18:
esp_mesh_lite_disconnect() will disconnect the current station connection and suspend internal reconnection mechanisms until esp_mesh_lite_connect() is called.