espressif/esp-mesh-lite #179: master 版本 esp_mesh_lite_set_mesh_id 导致崩溃重启

raspi404, 2025-07-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

None

### Actual behavior (suspected bug)

[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

```c
esp_mesh_lite_set_mesh_id(mesh_config->mesh_id,` true);
esp_mesh_lite_init(&mesh_lite_config);
```

### Project release version

v1.0.2 - 2025-6-27

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

esp32 c3

### Shell

Bash

### Additional context

不使用 `esp_mesh_lite_set_mesh_id` ，没办法在不清 flash 的情况下成功更新 `mesh id` 
最新的 `mester` 版本，使用 `esp_mesh_lite_set_mesh_id` 会导致崩溃重启
在 `v1.0.2 - 2025-6-27` 之前的版本都没有问题

> 另外为什么`esp_mesh_lite_init`不一定接受参数里的值去init？

tswen, 2025-07-02:
esp_mesh_lite_set_mesh_id 不可以在 esp_mesh_lite_init 之前调用。

raspi404, 2025-07-02:
> esp_mesh_lite_set_mesh_id 不可以在 esp_mesh_lite_init 之前调用。

上个版本我是这样用的，改在之后没有问题了。
为什么esp_mesh_lite_init不一定接受参数里的值去init？

closed 2025-07-02 by raspi404

tswen, 2025-07-02:
在没存入 nvs 中之前，是可以根据 init 里面的参数去设置的。一旦存入 nvs，init 都是会按照 nvs 中的 mesh id 值

raspi404, 2025-07-02:
> 在没存入 nvs 中之前，是可以根据 init 里面的参数去设置的。一旦存入 nvs，init 都是会按照 nvs 中的 mesh id 值

这个api每次开机都调用一次，但生命周期长的老配置比生命周期短的新配置有更高的优先级，传入的新配置不能覆盖老配置，这样设计的应用场景是什么？
vendor_id 会不会也受此影响？