espressif/esp-mesh-lite #180: 最新 master 特定情况两个不同 id 的 mesh 节点非法互通

raspi404, 2025-07-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

None

### Actual behavior (suspected bug)

这个是 id 为 71 的测试 mesh
[image]

这个是 id 为 172 的项目 mesh
[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

```c
uint8_t root_node_mac[6] = {0xdc, 0x06, 0x75, 0xf8, 0xa1, 0xe0}; // {0xdc, 0x06, 0x75, 0xf8, 0xb4, 0xa8};

bool is_mesh_root = false;

void mesh_lite_init(void)
{
    static char softap_ssid[32] = {0};

    snprintf(softap_ssid, sizeof(softap_ssid), "%.31s", CONFIG_BRIDGE_SOFTAP_SSID);

    esp_mesh_lite_set_softap_info(softap_ssid, CONFIG_BRIDGE_SOFTAP_PASSWORD);

    wifi_config_t wifi_config = {0};
    esp_bridge_wifi_set_config(WIFI_IF_STA, &wifi_config);
    snprintf((char *)wifi_config.ap.ssid, sizeof(wifi_config.ap.ssid), "%s", softap_ssid);
    strlcpy((char *)wifi_config.ap.password, CONFIG_BRIDGE_SOFTAP_PASSWORD,
            sizeof(wifi_config.ap.password));

    wifi_config.ap.channel = 0;

    ESP_ERROR_CHECK(esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_config));

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();

    mesh_lite_config.join_mesh_ignore_router_status = true;
    mesh_lite_config.join_mesh_without_configured_wifi = !is_mesh_root;

    esp_mesh_lite_init(&mesh_lite_config);
}
```

### Project release version

v1.0.2 - 2025-7-2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

esp32 c3

### Shell

CMD

### Additional context

测试 mesh 还是沿用 [https://github.com/espressif/esp-mesh-lite/issues/174](https://github.com/espressif/esp-mesh-lite/issues/174) 的测试项目，并且更新为 `v1.0.2 - 2025-7-2` 

测试 mesh 有一个根节点，一个子节点，都是是原本项目 mesh 中的子节点，重新通过 `idf.py -p COM17 erase-flash flash monitor` 烧录得来

在项目 mesh 没有所有节点重启的情况下，即使测试 mesh 有不同的 id 有自己的配置上的根节点，也会连接到项目 mesh 成为子节点。

tswen, 2025-07-18:
1. 烧录 1 个 no router 示例的根节点并上电（meshid 77）
2. 烧录 3 个 no router 示例的子节点并上电（meshid 77）
3. 修改其中一个子节点的 mesh id 为 1，烧录重新上电（不能通过 menuconfig 设置，如果一定要通过 menuconfig 设置，请先 idf.py erase-flash，因为第一次上电 77 已经被存入 nvs 中；或者你可以在新的烧录固件中 esp_mesh_lite_init 之后通过 esp_mesh_lite_set_mesh_id 设置新的 mesh id）
4. 步骤 3 上电的子节点设备并不会连接到 步骤 1 上电的根节点上

请注意第三条括号中的内容，另外日志在 mesh lite 初始化中也打印了 mesh id 的值，您是否没有正确的设置另一个 mesh id？
另外有一个小建议，您可以定期通过 esp_mesh_lite_get_mesh_id 打印下 mesh id 的值，以帮您更好的确认问题。

raspi404, 2025-07-18:
> 1. 烧录 1 个 no router 示例的根节点并上电（meshid 77）
> 2. 烧录 3 个 no router 示例的子节点并上电（meshid 77）
> 3. 修改其中一个子节点的 mesh id 为 1，烧录重新上电（不能通过 menuconfig 设置，如果一定要通过 menuconfig 设置，请先 idf.py erase-flash，因为第一次上电 77 已经被存入 nvs [...]

我测试案例是步骤 1 原本有一个根节点六个子节点，把其中两个子节点通过 `idf.py -p COMxx erase-flash flash monitor` 烧录另一个 mesh 配置的项目（其中一个根节点一个子节点），但是这两个节点上电后还是连接 1 所在的网络

单重启步骤 2 中的那两个节点不能解决问题，重启步骤 1 中剩余的五个节点，可以让步骤 2 中的两个节点重新连回自己所在的 mesh

这个过程我在发 issues 的时候测试了两次以上