espressif/esp-mesh-lite #182: espnow last version & meshlite v1.0.2 discussion and questions

yel-best, 2025-07-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

use meshlite v1.0.2

* meshlite default, the ESPNOW service is enabled if I follow the instructions directly https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide.md#espnow-usage-guide  Will using espnow support tandem relaying like meshlite? So far I've tested ESPNOW broadcasts in a relatively clean environment using LR for up to 500m, but can I add two esp32s in the middle for 1,000m-1,500m?
* meshlite doesn't seem to need to initialize espnow just register to send events & send msg, right? do we need to make a match?

thank you for your reply 😄

tswen, 2025-07-11:
- By default, ESP-NOW does not support message relaying functionality; the application layer needs to implement the relaying logic on its own.

- Yes, you just need to register the corresponding message type and receive callback interface, then call the provided send function. Currently, before sending, you need to add the peer device via esp_now_add_peer.

raspi404, 2025-07-11:
> * By default, ESP-NOW does not support message relaying functionality; the application layer needs to implement the relaying logic on its own.
> * Yes, you just need to register the corresponding [...]

Does that mean that even within the same MESH network, communication between devices using MESH LITE's customized ESPNOW might still fail due to the lack of message relaying support in ESPNOW? Would using TCP communication avoid this kind of issue?

tswen, 2025-07-18:
1. ESP-NOW message relaying can be implemented at the application layer.  
2. TCP only enables direct communication between devices under the same branch; devices across different branches still require forwarding by parent nodes.