espressif/esp-mesh-lite #183: Function esp_mesh_lite_transmit_file_start broken between v1.0.1 and v1.0.2

mrjohannchang, 2025-07-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Our product relies heavily on the ESP-Mesh-Lite library. Each deployed system typically includes **10 to 30 nodes**, and we use the built-in **LAN OTA update feature**, specifically the `esp_mesh_lite_transmit_file_start()` function, to efficiently roll out firmware updates across the mesh network.

We recently upgraded the library from **v1.0.1** (commit [87559f6](https://github.com/espressif/esp-mesh-lite/commit/87559f627e17313f2113403c62582825d335bd19)) to **v1.0.2**, and discovered that **LAN OTA breaks when a parent node is running v1.0.2 and its child nodes are still on v1.0.1**.

We suspect this issue stems from the **breaking changes to the** `get_file_cb` **callback**, as outlined in the [changelog](https://github.com/espressif/esp-mesh-lite/blob/da20046fb3b0421881b0143d4dc130915d3bc952/components/mesh_lite/CHANGELOG.md?plain=1#L9C3-L9C24), and likely introduced in commit [59f36a1](https://github.com/espressif/esp-mesh-lite/commit/59f36a150da9b0fd6154ab2361a781ce1187b537).

## Impact

We have **thousands of field-deployed devices** currently running firmware based on v1.0.1. If `esp_mesh_lite_transmit_file_start()` is no longer backward-compatible between v1.0.2 and v1.0.1, updating each device via HTTP OTA individually would be **extremely labor-intensive and impractical**.

## Request

* Can you please confirm whether **backward compatibility between v1.0.2 and v1.0.1** for esp_mesh_lite_transmit_file_start() is supported?
* If not, is there a **recommended workaround** for performing a **version bridge or phased upgrade** via LAN OTA (e.g. an intermediate compatibility layer)?
* Alternatively, is there a patch or branch available that restores v1.0.1 compatibility for the OTA flow?

We greatly appreciate your help and would welcome any guidance to help resolve this issue efficiently. Thank you!

mrjohannchang, 2025-07-22:
@tswen, want to have you attention here.

mrjohannchang, 2025-07-30:
Hi @tswen,

After some trial and error, we’ve determined that the **actual breaking commit** is https://github.com/espressif/esp-mesh-lite/commit/70f5657e91e065546f2e626b8a4042de0a948f93.
—not https://github.com/espressif/esp-mesh-lite/commit/59f36a150da9b0fd6154ab2361a781ce1187b537 as we previously suspected.

We found that commit https://github.com/espressif/esp-mesh-lite/commit/0367d760c432699091aaa3d74e30d0aca601e3a0, which includes [the breaking get_file_cb change](https://github.com/espressif/esp-mesh-lite/commit/62655031e5dc527d35ebb2c51389f100208991cd) mentioned in the [changelog](https://github.com/espressif/esp-mesh-lite/blob/da20046fb3b0421881b0143d4dc130915d3bc952/components/mesh_lite/CHANGELOG.md#breaking-change), is **still compatible with v1.0.1** in our tests.

To isolate the issue further, we suspected that changes to `OTA_DATA_LEN` and `OTA_WND_DEFAULT` might be involved. So we modified the following values in menuconfig under commit https://github.com/espressif/esp-mesh-lite/commit/0367d760c432699091aaa3d74e30d0aca601e3a0:

- `OTA_DATA_LEN`: changed from **1388** to **1376**
- `OTA_WND_DEFAULT`: changed from **8328** to **8256**

However, after applying these changes, the **root device failed to connect to the router**, so we couldn’t test the impact on LAN OTA compatibility.

tswen, 2025-08-05:
Apologies, I missed a point earlier. When upgrading from v1 to v2, the ESP32 should not restart after completing the upgrade (CONFIG_AUTO_RESTART should be disabled). Then, the next-layer device needs to be notified to start executing esp_mesh_lite_transmit_file_start. Only after all devices have been upgraded to v2 firmware should they restart. Otherwise, if the parent node restarts immediately after upgrading, the new firmware's OTA_DATA_LEN and OTA_WND_DEFAULT values will change, which could still cause issues.  

This process may require storing the new firmware in a specific partition and using esp_mesh_lite_ota_register_file_transfer_cb to register provide_file_cb and get_file_cb, ensuring devices fetch the new firmware data from the correct partition.

mrjohannchang, 2025-08-08:
Thank you, @tswen.

If I understand correctly, we don’t actually need to tweak `OTA_DATA_LEN` and `OTA_WND_DEFAULT` in the bridge firmware. Instead, we just need to:

1. Build the bridge firmware based on v1.0.1.
2. Retrieve the firmware from the parent’s inactive partition, where the new v1.0.2-based firmware will be written, during LAN OTA downloading.
3. Ensure all nodes reboot at the same time after all finishes downloading

Could you confirm the procedure?

tswen, 2025-08-15:
Yes, you are correct. The key point is to ensure that ​OTA_DATA_LEN​ and ​OTA_WND_DEFAULT​ remain consistent between all devices requesting firmware and those providing firmware during ​LAN OTA.